OpenEIT/OpenEIT #2: Mock Data Mode

sebastianriese, 2017-12-28:
A mock serial handler that inputs recorded raw data (e.g. looped) would be nice to allow hacking on the software without needing a hardware setup. Further a library of recorded input data would facilitate the development of advanced reconstruction methods.

(Note: I would prefer if the recorded data would not be supplied in the github repository of the software but for example as a separate tar-ball that can be downloaded from the project website).

JeanRintoul, 2017-12-29:
I will upload some example data files when I get back home which are labelled, and can be read back in, or used for improving reconstruction algorithm.

sebastianriese, 2018-01-04:
I realized by now, that there *is* a mode to use recorded data, sorry for missing this at first. But more well labelled data would indeed be very nice.

JeanRintoul, 2018-01-08:
[anticlockwise_motion.txt](https://github.com/OpenEIT/EIT_Dashboard/files/1613298/anticlockwise_motion.txt)
yep, rename this as a .bin for now. I'll put it on dashboard repo after current pull request from your work is merged.

JeanRintoul, 2018-03-23:
added into repository.

closed 2018-03-23 by JeanRintoul