OpenEIT/OpenEIT #3: Current use of multiprocessing is bogus

sebastianriese, 2017-12-28:
The current use of the multiprocessing library is, as far as I can see, totally wrong. It should either be used not at all or used correctly, although I am not sure multiprocessing is the right fit, since we need to transport not so little data between the workers (raw data and image data), so threading or explicit daemons using unix sockets for communication might be a better choice.

JeanRintoul, 2017-12-29:
Agreed it is not currently functional, and could be completely removed. The thinking was that a threaded GUI would block the serial stream when intensive rendering took place due to Python's GIL(only one thread can run at once so they are blocking too, but having multiple processes running on different cores is non-blocking to the serial port) - inspired by this article on threading and the GIL: http://www.dabeaz.com/python/UnderstandingGIL.pdf ... At this stage, rendering is the most intensive part of the process, it would be good to either implement threading that guarantees no serial block, or multiprocessing the correct way.

JeanRintoul, 2018-03-23:
updated in major refactor which is now merged.

closed 2018-03-23 by JeanRintoul