OpenEIT/OpenEIT #5: Defining Internal Interfaces

sebastianriese, 2018-01-03:
The framework should support abstractions, that can be used to write code that works interchangeably with slightly different hardware (e. g. number and configuration of the electrodes, different wire protocol on the serial bus), and different reconstruction algorithms.

JeanRintoul, 2018-01-08:
Agreed in general although the different combinations are hard to predict to abstract away. Here are a few that are certain changes: different multi-frequency protocols, number of electrode and position of electrode changes(the precise position must be known), timing/sample rate changes, hardware revisions, more reconstruction algorithms.

JeanRintoul, 2018-04-13:
I've now started a configuration file for defining internal interfaces for different hardware/software. Also the new algorithm wrapper extensions enable different algorithm configurations.

closed 2018-04-13 by JeanRintoul