OpenEIT/OpenEIT #6: Replacing tkinter and matplotlib?

sebastianriese, 2018-01-04:
While using tkinter and matplotlib has some advantages (accessibility, availability, simplicity, etc.), it also has some drawbacks:
* parallelism and tkinter don't play along nicely, you don't get around polling for new data, if it is calculated in parallel.
* tkinter has no UI generator which makes changes to the UI complicated and error prone.
* while matplotlib allows simple plotting of the data and makes it easy to add histograms and other derived plots it is quite slow (plotting a 100x100 array should not take 40ms).

My preferred alternative would be to use Qt for the interface and OpenGL for plotting. OpenGL for plotting the reconstruction is feasibly since this only requires drawing an array with a color-map (which is possible with just a few lines of OpenGL and shader code). Using a graphics library like cairo is feasible as well.

Rendering of the data might be not too complicated with OpenGL (used as a 2D graphics library). Using OpenGL would also allow rendering data obtained from finite elements methods quite directly (especially when piecewise-linear basis functions are used – I have a simple proof of concept for this) and very fast. The color-map for pixel-based reconstructions could then be applied on the fragment shader. Also, integrating OpenGL with Qt is simple.

JeanRintoul, 2018-01-08:
Yes the rendering time is a major issue and it would be good to optimise this before more complex reconstruction techniques get integrated which will take more time such as EIDORS or pyEIT libraries. I see no problem with moving to OpenGL shader implementation, as long as it is easy to install(i.e. just installing anaconda is nice as it configures everything people need to run). If people want to use matplotlib for separate plot testing obviously they can still import and do it in their own sandbox.

sebastianriese, 2018-01-17:
Shipping Cython or C code should not be necessary. Anaconda (and probably all Linux distros, etc.) contain a PyQt5 package (and if needed PyOpenGL, but PyQt includes an OpenGL wrapper, so PyOpenGL should not even necessary). This means we can do everything we need with pure Python using the PyQt5 package.

PyQt5 package for anaconda: <https://anaconda.org/anaconda/pyqt>

JeanRintoul, 2018-01-23:
Qt sounds great. I hadn't realised PyQt has an OpenGL wrapper.

JeanRintoul, 2018-09-11:
It seems like electron and plotly may be a good alternative to solve this problem. Start of example located in sandbox folder.

JeanRintoul, 2018-11-30:
This solution has now been implemented. The app now runs a dash-plotly frontend, that can be run on a web browser on any platform, or using an installer to run as a standalone application without any extra installs.

closed 2018-11-30 by JeanRintoul