OpenEIT/OpenEIT #7: Another Reconstruction Algorithm

JeanRintoul, 2018-01-08:
It would be excellent to have another reconstruction algorithm apart from the already existing linear back-projection. This link to an open source project sounds promising if someone would like to adapt it into a module that can be used by the open source python dashboard: 
https://github.com/liubenyuan/pyEIT

Otherwise there is EIDORS(http://eidors3d.sourceforge.net/)  which is unfortunately in matlab(costs $$ outside of academia) where algorithms could be repurposed.  Also, of interest is the ASTRA toolbox: http://www.astra-toolbox.com/

It would be great to get a framework started to test different algorithms.

odo2063, 2018-01-11:
EIDORS, also works with GNU Octave which is free and usable with oct2py package...see my GUMS repo for example...

odo2063, 2018-01-11:
Doing something better on our selfs would be more fun while waiting for some PCBs...XD

JeanRintoul, 2018-01-16:
What about pyEIT? Does that one look good as it just came out? 
Sebastian has done some nice GUI efficiency updates that should be merged soon, that'll make it easier to integrate different types of reconstructions. I'm nervous about how much time FEM will take, do you think it can run in real-time?

odo2063, 2018-01-17:
I had brief look into pyEIT...and i dont like it XD. they are using to much abbreviations so (at least) I don't know what they are doing in there code...

liubenyuan, 2018-04-10:
@odo2063 sorry to hear that XD. If you have any suggestion, you may start an issue and pull a commit.

Currently, `pyEIT` does not support CEM models compared to EIDORS. It has almost the basic reconstruction algorithms alike EIDORS. `pyEIT` may be better integrated into your python GUI.

Thanks for your open EIT project. It looks interesting.

JeanRintoul, 2018-04-13:
pyEIT is convenient as it's in python(and is highly influenced by EIDORS algorithms). I've now integrated it into the dashboard, configurable through a config file. If you would prefer to use EIDORS, since it is in Matlab, you still can just as easily as before.

closed 2018-04-13 by JeanRintoul