OpenEIT/OpenEIT #8: How about defining a greater goal?

odo2063, 2018-01-11:
It should be a greater Goal defined to have a vague direction everybody can walk in.

My suggestion, measure the size of the seeds of an apple.

JeanRintoul, 2018-01-16:
Sure - it's worth concentrating on Spatial Resolution for a while - this is great. An apple is easily accessible as well. This issues list is for the Python Dashboard, where GUI, serial and reconstruction algorithms can be worked on. The goal of all the reconstruction algorithms is to obtain higher spatial resolution, though different physical set ups will also help with this. 

For improved spatial resolution:
1. Improved mathematical reconstructions. 
2. A new ME repository, which encompasses a range of 3D printable physical set ups, potentially including stepper motors for rotation of a tank etc. This will be easier to do once people can have PCB kits and I'm working on that now. There will be breakouts on the board for I2C/UART/SPI additions for things like stepper motor control. 
3. Adjusting EE constraints such as adding tetrapolar differential arrangement(I'm on this as soon as I hit the return key to this comment). 

Please hold tight and have patience as this is definitely a long term project - presentation at 34C3 was first mention of this work, and I'm excited to see interest and also more collaborators to work with in open biomedical imaging.

odo2063, 2018-01-17:
I am up to design a moduler PCB for the chips i showed you. So we can have up to 64 Channels per I²C-Bus with optional inputfilters(DC-Block, highpass) and optional inputrectifier(no need for FFT anymore and one value per channel is enough). this i would call "Whiteboard". next step in mind would be "Redboard", maybe different ADC und multiplexing for I²C and ADC channels so maybe up to 4096 channels per I²C would be possible. this will not be fast but should give a nice resolution. "Blackboard" could be with maxim pixi chips which have selectable ADC or DAC on upto 20 pins(i think)...but we will see along the way...

closed 2018-03-23 by JeanRintoul