OpenEIT/OpenEIT #13: Sorry the win32 platform is not supported by the BLE library!

Sifat-Hasan-3003, 2018-10-01:
I'm a windows user. When I tried to install dashboard.py it caused that problem. Is there any solution for that? Can I connect an EIT circuit with EIT_Dashboard via USB cable?

JeanRintoul, 2018-10-01:
You can connect with the USB cable on the next upcoming hardware revision which is finished at the fab, and awaiting shipment and testing. 

Otherwise, would love a contribution to make the bluetooth work in in Windows! (I try to use OSX most of the time, and I think the bluetooth library also works for Linux).

JeanRintoul, 2019-03-25:
Linux and OSX are supported for BLE, but Windows is not currently. Will add Windows BLE support on a future revision.

JeanRintoul, 2019-04-24:
The best all OS solution to BLE I have found is this library but it appears early(something to keep an eye on though!): https://github.com/hbldh/bleak/

Otherwise, you can always plug the device in via the USB cable to transmit data via standard 115.2K UART COMMS protocol.

closed 2020-09-03 by JeanRintoul