OpenEIT/OpenEIT #19: KeyError: 'map'

ickkim, 2019-03-24:
Hi. 
Would you let me know what went wrong?
I also downloaded you dmg file and got stuck similarly.
Thanks a lot in advance.

Traceback (most recent call last):
  File "/Users/sungick/tensorflow/lib/python3.6/site-packages/flask/app.py", line 2309, in __call__
    return self.wsgi_app(environ, start_response)
  File "/Users/sungick/tensorflow/lib/python3.6/site-packages/flask/app.py", line 2295, in wsgi_app
    response = self.handle_exception(e)
  File "/Users/sungick/tensorflow/lib/python3.6/site-packages/flask/app.py", line 1741, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/Users/sungick/tensorflow/lib/python3.6/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/Users/sungick/tensorflow/lib/python3.6/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/Users/sungick/tensorflow/lib/python3.6/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/Users/sungick/tensorflow/lib/python3.6/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/Users/sungick/tensorflow/lib/python3.6/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/Users/sungick/tensorflow/lib/python3.6/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/Users/sungick/tensorflow/lib/python3.6/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/Users/sungick/tensorflow/lib/python3.6/site-packages/dash/dash.py", line 450, in serve_component_suites
    })[path_in_package_dist.split('.')[-1]]
KeyError: 'map'

JeanRintoul, 2019-03-25:
Great - thanks for reporting. I just tried this and there seems to be an operating system/dash version/python version dependency error. I put some new install instructions in the readme so please give those a go. 

First, I updated to Python 3.6.7 like this: 
`conda update --all`

Then check the version by: `python --version `
To ensure you are installing in a fresh environment create a virtual environment like this: 

```
virtualenv -p python3 envtest

source envtest/bin/activate

git clone https://github.com/OpenEIT/OpenEIT.git

cd OpenEIT

pip install -r requirements.txt

python app.py 
```

Bourbon-Whiskey, 2019-04-24:
I also want to know if you have succeeded in running later?

JeanRintoul, 2019-04-24:
@Bourbon-Whiskey There was a pull request that was just merged 3 days ago for Ubuntu 18.04. Do you have this latest update? 

This problem is certainly an install dependency problem. Are you running in a virtual environment as described above(good way to check you don't have a dependency issue)? 

Otherwise, could I get a report on the versions of the libraries you are running? 
i.e. pip show imageio etc for all versions of libraries you have installed(possibly a library did install correctly during the install procedure)?

JeanRintoul, 2019-04-27:
Please pull and try again. Can confirm operation on 3 operating systems.