OpenEIT/OpenEIT #21: Why non commercial?

atrent, 2019-04-07:
w.r.t. the license.

(this is a major obstacle for me to back your project)

JeanRintoul, 2019-04-07:
Spectra is sold at the most cost effective price ever for an EIT system so far.

The plan with the license is it's a the non-commercial share alike for people who want to try and experiment, hack, improve or academics who do research and publish - meaning the updates stay in the open to make a better commons for everyone. By leaving the option for commercial licenses open for users who keep their updates private, we hope to make a way to sustain the project in a reasonable way, so we can keep working on it and spend time maintaining it which should help any private businesses also by keeping it reliable and up to date with the latest technological improvements. Basically seeking a financial symbol of gratitude for the work done from people who create profitable businesses based on it. 

Hope you still consider backing the project! I will be posting a series of updates on the science behind it, and different demonstrations of what you can do with it on the crowd supply updates mailing list over the coming weeks.

atrent, 2019-04-07:
I'd like to "+1" to this RFE.
A way to save board (and related settings) parsable info into the code or as a side file would be very useful.

JeanRintoul, 2019-04-07:
@atrent agreed. Perhaps this could be a future update!

atrent, 2019-04-07:
sorry, I posted a comment in the wrong window

atrent, 2019-04-07:
I'm afraid I cannot back your project (because of your not-entirely-free license) but I will follow the progress since I find it very interesting.

closed 2019-04-07 by JeanRintoul