OpenEIT/OpenEIT #24: Request for help

Bourbon-Whiskey, 2019-04-24:
Hello, I have some questions about this project that I have to ask：
Since I started trying to run this project on my laptop a week ago, I tried windows (win10_x64) and Ubuntu (18.10). On both platforms, I used Anaconda according to your readme.md requirements and configured python3.6.7. The environment has all the required packages installed, but the project does not work on both platforms. Did I miss something?
I look forward to seeing you help me find out where the problem is.Thankyou.

The window error message is as follows (I put the item in "E:/" and replace the time with x):
`windows users
xxxx-xx-xx xx:xx:xx,xxx JAC mesh set up
completed jac reset
xxxx-xx-xx xx:xx:xx,xxx JAC mesh set up
completed jac reset
d
Traceback (most recent call last):
  File "app.py", line 77, in <module>
    main()
  File "app.py", line 71, in main
    gui = runGui(controller)
  File "E:\OpenEIT\OpenEIT\dashboard\dash_control.py", line 61, in __init__
    self.fw_display = fw.FWgui(self.controller,self.app)
  File "E:\OpenEIT\OpenEIT\dashboard\modes\fw.py", line 60, in __init__
    rest = str(full_ports[-1]).split(sep, 1)[0]
IndexError: list index out of range`
The Ubuntu error message is as follows (I put the item in Home folder and replace the time and my username with x):
`xxxx-xx-xx xx:xx:xx,xxx JAC mesh set up 
completed jac reset
xxxx-xx-xx xx:xx:xx,xxx JAC mesh set up 
completed jac reset
d
xxxx-xx-xx xx:xx:xx,xxx openeit_server_started
Running on http://127.0.0.1:8050/
xxxx-xx-xx xx:xx:xx,xxx Running on http://127.0.0.1:8050/
Debugger PIN: 853-637-018
xxxx-xx-xx xx:xx:xx,xxx Debugger PIN: 853-637-018
 * Serving Flask app "app" (lazy loading)
 * Environment: development
 * Debug mode: on
Traceback (most recent call last):
  File "app.py", line 79, in <module>
    main()
  File "app.py", line 74, in main
    gui.run()
  File "/home/xxxx/OpenEIT/OpenEIT/dashboard/dash_control.py", line 218, in run
    self.app.run_server(debug=True)
  File "/home/xxxx/anaconda3/lib/python3.6/site-packages/dash/dash.py", line 1292, in run_server
    **flask_run_options)
  File "/home/xxxx/anaconda3/lib/python3.6/site-packages/flask/app.py", line 943, in run
    run_simple(host, port, self, **options)
  File "/home/xxxx/anaconda3/lib/python3.6/site-packages/werkzeug/serving.py", line 988, in run_simple
    run_with_reloader(inner, extra_files, reloader_interval, reloader_type)
  File "/home/xxxxx/anaconda3/lib/python3.6/site-packages/werkzeug/_reloader.py", line 332, in run_with_reloader
    sys.exit(reloader.restart_with_reloader())
  File "/home/xxxx/anaconda3/lib/python3.6/site-packages/werkzeug/_reloader.py", line 176, in restart_with_reloader
    exit_code = subprocess.call(args, env=new_environ, close_fds=False)
  File "/home/xxxx/anaconda3/lib/python3.6/subprocess.py", line 267, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/home/xxxx/anaconda3/lib/python3.6/subprocess.py", line 709, in __init__
    restore_signals, start_new_session)
  File "/home/xxxx/anaconda3/lib/python3.6/subprocess.py", line 1344, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 8] Exec format error: '/home/xxxx/OpenEIT/app.py'`

JeanRintoul, 2019-04-24:
@Bourbon-Whiskey There was a pull request that was just merged 3 days ago for Ubuntu 18.04. Do you have this latest update?

This problem is certainly an install dependency problem. Are you running in a virtual environment(good way to check you don't have a dependency issue)?

To start from a clean installation I'd suggest the following instructions which include a virtual environment just to make sure you have all the right versions of libraries installed - this would also give as a good equal base place to start on our phone call: 

First, I updated to Python 3.6.7 like this:
conda update --all

Then check the version by: python --version 
To ensure you are installing in a fresh environment create a virtual environment like this:

virtualenv -p python3 envtest

source envtest/bin/activate

git clone https://github.com/OpenEIT/OpenEIT.git

cd OpenEIT

pip install -r requirements.txt

python app.py 
Let me know if this now works for you.

Otherwise, could I get a report on the versions of the libraries you are running?
i.e. pip show imageio etc for all versions of libraries you have installed(possibly a library did install correctly during the install procedure)?

JeanRintoul, 2019-04-24:
another thing you can do is cd into your OpenEIT directory and type `pip check`

Bourbon-Whiskey, 2019-04-25:
Wow,amazing!
After I reconfigure the environment with the method you provided, the IDE does not prompt the bug anymore. According to the cached package status duiring the pip, I lacked the package named "certifi" before. Reconfiguring the python environment did solve the problem. All packages are listed below according to your requirements（Python version 3.6.8）.

> Adafruit-BluefruitLE==0.9.10
attrs==19.1.0
certifi==2019.3.9
chardet==3.0.4
Click==7.0
cloudpickle==0.8.1
cycler==0.10.0
dash==0.35.0
dash-core-components==0.42.1
dash-html-components==0.13.5
dash-renderer==0.22.0
dask==1.2.0
decorator==4.4.0
Flask==1.0.2
Flask-Compress==1.4.0
future==0.17.1
idna==2.8
imageio==2.4.1
ipython-genutils==0.2.0
itsdangerous==1.1.0
Jinja2==2.10.1
jsonschema==3.0.1
jupyter-core==4.4.0
MarkupSafe==1.1.1
matplotlib==2.1.1
nbformat==4.4.0
networkx==2.3
numpy==1.16.2
Pillow==6.0.0
plotly==3.8.1
pyparsing==2.4.0
pyrsistent==0.14.11
pyserial==3.4
python-dateutil==2.8.0
pytz==2019.1
PyWavelets==1.0.3
requests==2.21.0
retrying==1.3.3
scikit-image==0.14.2
scipy==1.2.1
six==1.11.0
toolz==0.9.0
traitlets==4.3.2
urllib3==1.24.2
Werkzeug==0.15.2

There is a new problem. After running `python app.py`, it stucked for a long time. Theterminal shown as follows ( time is hidden again):
```
xxxx-xx-xx xx:xx:xx,xxx JAC mesh set up 
completed jac reset
xxxx-xx-xx xx:xx:xx,xxx JAC mesh set up 
completed jac reset
d
xxxx-xx-xx xx:xx:xx,xxx openeit_server_started
Running on http://127.0.0.1:8050/
xxxx-xx-xx xx:xx:xx,xxx Running on http://127.0.0.1:8050/
Debugger PIN: 257-959-416
xxxx-xx-xx xx:xx:xx,xxx Debugger PIN: 257-959-416
 * Serving Flask app "app" (lazy loading)
 * Environment: development
 * Debug mode: on
xxxx-xx-xx xx:xx:xx,xxx JAC mesh set up 
completed jac reset
xxxx-xx-xx xx:xx:xx,xxx JAC mesh set up 
completed jac reset
d
xxxx-xx-xx xx:xx:xx,xxx openeit_server_started
Running on http://127.0.0.1:8050/
xxxx-xx-xx xx:xx:xx,xxx Running on http://127.0.0.1:8050/
Debugger PIN: 686-510-736
xxxx-xx-xx xx:xx:xx,xxx Debugger PIN: 686-510-736

```

After opening [http://127.0.0.1/8050](url), there is a page with Title"Dash"and only "Loading.." in its body for quite a long time.An hour has passed and nothing has changed.

Is this normal?Or what have I missed?

Thank you.

JeanRintoul, 2019-04-25:
It shouldn't be stuck for a long time indicating you still have some sort of install dependency bug the most obvious thing I can see is your version of python is 3.6.8 not 3.6.7. Can you try again with Python 3.6.7? Also Certifi is not one of OpenEIT's dependencies.

I've done testing only for python 3.6.7(on Windows and OSX) and John Naulty has successfully testing this works on Ubuntu 18.10).

Bourbon-Whiskey, 2019-04-26:
I have switched to python3.6.7 but the results have not changed. I used Ubuntu 18.10, Anaconda 2019.03 (latest version), and used Python 3.6.8 and 3.6.7.
Thank you for your help anyway.

Bourbon-Whiskey, 2019-04-26:
This is the result of my running after using Windows:
```
windows users
xxxx-xx-xx xx:xx:xx,xxx JAC mesh set up
completed jac reset
xxxx-xx-xx xx:xx:xx,xxx JAC mesh set up
completed jac reset
d
Traceback (most recent call last):
  File "app.py", line 77, in <module>
    main()
  File "app.py", line 71, in main
    gui = runGui(controller)
  File "E:\OpenEIT\OpenEIT\dashboard\dash_control.py", line 61, in __init__
    self.fw_display = fw.FWgui(self.controller,self.app)
  File "E:\OpenEIT\OpenEIT\dashboard\modes\fw.py", line 60, in __init__
    rest = str(full_ports[-1]).split(sep, 1)[0]
IndexError: list index out of range
```
I am using win10 (x64) Pro, using the latest version of Anaconda, python3.6.7, the package information is as follows:

> Adafruit-BluefruitLE==0.9.10
attrs==19.1.0
certifi==2019.3.9
chardet==3.0.4
Click==7.0
cloudpickle==0.8.1
cycler==0.10.0
dash==0.35.0
dash-core-components==0.42.1
dash-html-components==0.13.5
dash-renderer==0.23.0
dask==1.2.0
decorator==4.4.0
Flask==1.0.2
Flask-Compress==1.4.0
future==0.17.1
idna==2.8
imageio==2.4.1
ipython-genutils==0.2.0
itsdangerous==1.1.0
Jinja2==2.10.1
jsonschema==3.0.1
jupyter-core==4.4.0
MarkupSafe==1.1.1
matplotlib==2.1.1
nbformat==4.4.0
networkx==2.3
numpy==1.16.2
Pillow==6.0.0
plotly==3.8.1
pyparsing==2.4.0
pyrsistent==0.15.0
pyserial==3.4
python-dateutil==2.8.0
pytz==2019.1
PyWavelets==1.0.3
requests==2.21.0
retrying==1.3.3
scikit-image==0.14.2
scipy==1.2.1
six==1.11.0
toolz==0.9.0
traitlets==4.3.2
urllib3==1.24.2
Werkzeug==0.15.2
wincertstore==0.2

Thank you for your help anyway.

JeanRintoul, 2019-04-26:
I have been able to recreate your problem on Ubuntu which I just installed. It looks very much like a a dependency issue, I suspect it's the version of dash/plotly libraries on ubuntu. It is working for me on both windows and OSX(I just installed Ubuntu). 

I have tried running it on all 3 platforms in the following ways and succeeded: 
1. Windows 10: Installed it successfully with my default anaconda installation which is running Python 3.7.1 on windows. Not a problem and up and running. I did have to press the return/enter key a couple of times after entering `python app.py` into my command prompt. 
2. OSX: not a problem works great. 
3. Windows 10: I followed the virtualenv directions I gave above on Windows 10, the only difference is virtual env activate requires a different command on windows than OSX - to activate it follow these directions(https://programwithus.com/learn-to-code/Pip-and-virtualenv-on-Windows/). Then proceed as before. The dash started and works.  
4. I just installed Ubuntu 18.04.2: It hangs on load as you described. I bet it's related to the dash/plotly library versions. TBD.

JeanRintoul, 2019-04-27:
Ok, please do a new pull and it should run - I've updated the dash library in the requirements file and tested it on OSX/Windows and Ubuntu. This new update of dash comes with some handy new debugging info(which hasn't been completely integrated into OpenEIT yet, but can do that on another day).

Bourbon-Whiskey, 2019-04-29:
Sorry for let you wait for such a long time, I took a two-day weekend break without using my computer.

I updated the project in your method, and it did change: the page was displayed. However, problems related to dash framework still arise in the running process, whether in Ubuntu or Windows.

BTW, I found it interesting that installing 2.1.1 version of Matplotlib as required by your "requirements.txt" will be successful under Python 3.6, but by Python 3.7 you will need an updated version of Matplotlib to install successfully.

Anyway, I would like to thank you for your kindly response. I will try the OSX laptop in a few days, thank you.

JeanRintoul, 2020-09-03:
Any updates on this issue?

closed 2020-09-03 by JeanRintoul