OpenEIT/OpenEIT #26: Timeseries doesn't show data until after switching to Spectroscopy tab.

DanLipsitt, 2019-05-09:
To reproduce:

1. Connect (tested with USB).
1. Send `a` to microcontroller.
1. Switch to TimeSeries tab - the chart is blank.
1. Switch to Spectroscopy tab. The spectrogram renders and updates.
1. Switch back to TimeSeries tab. The timeseries renders and updates.

McD-NMI, 2019-11-15:
Having a similar issue. Timeseries just doesnt render. If I go to Spectroscopy tab it doesnt necesxsarily update. Sometimes if I am lucky then I see the timeseries plot but then it disappears if I switch tabs

JeanRintoul, 2020-09-03:
Please see video tutorial and change modes per this tutorial seen at openeit.github.io. I'd love any bug updates if anyone would like to attempt to resolve.

closed 2020-09-03 by JeanRintoul