OpenEIT/OpenEIT #29: Error in dashboard when changing algorithm settings from imaging tab

wjenh, 2019-10-08:
Problem in the dashboard - changing algorithm settings causes the imaging display to stop working. 32 electrode mode doesn't seem to work at all. I have to hard-reset the spectra to recover. Here's one stack trace: ValueError: all the input array dimensions except for the concatenation axis must match exactly
2019-10-07 08:30:35,469 Exception on /_dash-update-component [POST]
Traceback (most recent call last):
  File "D:\Python\lib\site-packages\flask\app.py", line 2446, in wsgi_app
    response = self.full_dispatch_request()
  File "D:\Python\lib\site-packages\flask\app.py", line 1951, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "D:\Python\lib\site-packages\flask\app.py", line 1820, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "D:\Python\lib\site-packages\flask_compat.py", line 39, in reraise
    raise value
  File "D:\Python\lib\site-packages\flask\app.py", line 1949, in full_dispatch_request
    rv = self.dispatch_request()
  File "D:\Python\lib\site-packages\flask\app.py", line 1935, in dispatch_request
    return self.view_functions[rule.endpoint](req.view_args)
  File "D:\Python\lib\site-packages\dash\dash.py", line 1147, in dispatch
    response.set_data(self.callback_map[output]['callback'](args))
  File "D:\Python\lib\site-packages\dash\dash.py", line 1033, in add_context
    output_value = func(args, kwargs)
  File "C:\Tmp\OpenEIT-master\OpenEIT\dashboard\modes\imaging.py", line 569, in update_graph_scatter
    points3D = np.vstack((self.x,self.y,self.img)).T
  File "D:\Python\lib\site-packages\numpy\core\shape_base.py", line 283, in vstack
    return _nx.concatenate([atleast_2d(_m) for _m in tup], 0)

JeanRintoul, 2019-10-12:
Agreed there is a bug when changing electrode numbers on the imaging screen. For now, please change modes between 8, 16 and 32 electrodes on the control tab of the dashboard until this bug on the imaging tab is fixed. 
The commands you can use are listed at the bottom of the control tab. i.e. A and then hit send will switch it to time series. Make sure you can see the new data format streaming through in the text box, then switch to the time series tab. If you want 32 electrode EIT then type 'e' and hit send, and wait until you see a 32 electrode(that's 896 combinations of electrodes) listing in the text box, then switch back over to the imaging screen. Don't forget to baseline, autoscale and use the range sliders to adjust contrast on the image too! 

[image]

JeanRintoul, 2020-09-03:
Any updates on this issue? Are you able to connect? Have you watched the detailed video tutorial on openeit.github.io? Hopefully this can resolve any connection issues.

zhjackie2001, 2020-11-05:
Hi, I would like to follow up on this thread. I am experiencing the same error in the imaging mode, there is indeed a bug when switching modes directly on the imaging tab. However, Jean's suggestion from the control tab is a work around. 

However, I also noticed that if the "autoscale" or "baseline" on the imaging tab is previously checked, and switching the electrodes at the control dashboard will result in the following error message, which won't produce any image reconstruction updates. 
[image]

Interestingly, starting fresh, the 32-electrode mode (selected from control tab) does work, given no previous imaging tab options have been selected. The same error occurs but the program continues to run and produces reconstructed images, albeit it takes time to produce an image, so be patient. 
[image]

They both share the same error, any thoughts on the source and solution to this?