OpenEIT/OpenEIT #30: No EIT image shown in dashboard

FKShi, 2019-10-09:
I use Windows and run OpenEIT wit anaconda. When I finally successfully connected the EIT kit to the dashboard (the EIT kit was connected to the electrode in the tank), no images were reconstructed, there was only a monochromatic circle on the right. It seems that the impedance measurement was not conducted at all.

[image]
[image]

JeanRintoul, 2019-10-12:
It looks like you have no serial connection. Each Spectra unit has been tested to ensure the COM port works before sending it out - so it's likely a config error. What OS are you using? Can you stream serial using real-term(skip the dashboard and just look at a serial stream)?

credensjusticiam, 2019-10-18:
i have this same problem,.the serial connection was establshed, and the data is streamed,. but the reconstruction image is not shown, only the graphic

[image]
[image]

JeanRintoul, 2019-10-18:
First, take the object out of your tank, then wait, and select 'baseline'. Is there water in your tank? Now put the object back in and wait to see an image. An easy way to see if the data is being streamed to the imaging tab is to not the histogram which shows the numerical values of the image reconstruction and their distribution. Are they changing as data comes in?

McD-NMI, 2019-10-30:
I am having the exactsame issue, on Windows 10. 

Histogram and EIT image were updating for a little bit, but then it stopped. Data is straming, but histogram and EIT no logner change.

JeanRintoul, 2019-10-30:
Interesting. Can you give me exact steps to recreate this issue, then can look into it? 

- In the meantime, try turning the device off, then on, then reconnecting the dashboard.

McD-NMI, 2019-10-30:
I feel like it is actually related to Issue #29. Seems to happenign when I change the # of electrodes in the imaging tab. I get the same error messages in the terminal. Perhaps it is the same for @FKShi and @credensjusticiam?

credensjusticiam, 2019-11-01:
my problem is not solved yet, @McD-NMI ,. i've been trying @JeanRintoul suggestion, but still not working though,.right now the data is sometime streamed and sometime it is not,
to Jean @JeanRintou:l maybe you could give use video instruction how to set up the device, especially in windows based OS

JeanRintoul, 2020-01-28:
Here is a new video on basic instructions from unpacking through to imaging. Hope it helps: [](https://youtu.be/Ssm7BgsbbWI)

Alby99, 2020-07-23:
@JeanRintoul I just got the device and it took me a while to set it up, but everything runs right now except I don't get the image in the screen. I sometimes get a monochromatic circle, like the one in the first comment in the thread, but most of the time I get an image like this
[image]

The histogram will update with the new data for a few iterations, and will then stop updating even though I can see in the terminal new data is being taken.

Any solutions to this?

JeanRintoul, 2020-07-24:
I would say there is something wrong with the serial streaming - in which case it's worth showing that you have that set up correctly, and working. 
Questions: 
1. What operating system are you running? 
2. Did you install the serial VCP drivers if you are running windows as per the installation readme: https://github.com/OpenEIT/OpenEIT
3. Can you see a serial stream if you open a serial streaming program like realterm? 
4. When you open the EIT Dashboard, and then connect to the serial stream on the Control stream, do you see numbers fill up in the window where it says 'streamed data'?
5. Staying on the control tab, can you change the modes? For instance, it starts in EIT 16 electrode mode, if you type C and hit send, does it change mode and you can see new serial data streamed? 

First, have you gone through this tutorial:https://youtu.be/Ssm7BgsbbWI 
Second, if you go back to the

JeanRintoul, 2020-09-03:
Any further questions, or troubles, please open a new issue. Otherwise, with above detailed instructions and videos I assume this issue is resolved.

closed 2020-09-03 by JeanRintoul