OpenEIT/OpenEIT #31: Issue with connector

Guruprasad-Rao, 2019-10-17:
Hi Jean
Thank you for providing spectra. 

1 I think I have a missing component in the flap connector module. Could you please check the image and let me know. I cannot insert the connecting cables.
2 I am using windows 10-64 bit OS. I have seen BLE devices is not supported, I have installed bleak but have not tested it yet. Can I connect the spectra directly to the USB port in PC and how to establish communication. Please inform me. Or the microUSB is only for power connection?
[image]

JeanRintoul, 2019-10-17:
Looks like everything is in the right place. This video of how to connect the flex PCB might help you: https://openeit.github.io/docs/html/faq.html

The USB can power the board, or recharge the battery if you connect one. The microUSB also send serial data through it so you can connect directly to the board via UART. 

Let us know if you have further issues.

Guruprasad-Rao, 2019-10-17:
Thank you Jean

I have connected the spectra to PC. There was some error for plotly which was resolved after downgrading plotly==3.10.  I am trying to image using water and bottle as test. 
the default reconstructions were not clear but I need to learn more about it.
The raw data was acquired using 8, 16 and 32 electrode configuration. 
The time for 32 electrodes acquiring 896 readings is 35-36 seconds. This is a bit slow for me.

I am sorry I cannot understand communications properly so I have a question. Is the speed slow because currently the connection (when I press connect in dashboard) is via UART. How can I make a faster acquisition for 32 electrode mode. 

Also how is the data arranged, for reconstruction purpose. How to find out the excitation node and measuring node to perform reconstruction separately in eidors.

Regards

JeanRintoul, 2019-10-17:
This projects only supports the module the versions described in the requirements.txt folder. 

The communications are via UART if you are using Windows, and there is a stim pattern of 896 combinations. You can use simdata.py to simulation the same data patterns without the device to learn more about it. 

I suggest you baseline your phantom before imaging, and then use the contrast adjustment sliders and autoscale. A new video will arrive based on this in the next few weeks once its ready for everyone to acquire better images.

Guruprasad-Rao, 2019-10-17:
ok. Thank you very much.
I would be looking forward for the video. 
Meanwhile I would try the simdata also.