OpenEIT/OpenEIT #32: Adjusting frequency

fajartimur, 2019-10-17:
Hi Jean, 
I want to ask about certain topic, i read at this site "https://www.crowdsupply.com/mindseye-biomedical/spectra" it said that".....and electrical impedance tomography, between 80 Hz and 80 kHz, with up to 32 electrodes.". 
can you give me the instruction or tutorial how to adjust this frequency please?,.ie: i want to inject 75kHz signal (single frequency) at the phantom.
thank you

JeanRintoul, 2019-10-17:
Adjusting the frequency can be done in firmware - please check the firmware repositories.

fajartimur, 2019-10-17:
thank you for the reply,.where can i find this firmware repositories?, can you give a link please

JeanRintoul, 2019-10-17:
here is the top level listing all the directories of the project: https://github.com/openeit 

I think you can guess which one is firmware.

AdrienFoueillassar, 2020-02-11:
The frequency and amplitude of the injected signal are specified in the modes.h file

JeanRintoul, 2020-09-03:
Thanks @AdrienFoueillassar

closed 2020-09-03 by JeanRintoul