OpenEIT/OpenEIT #34: offline test data

orgua, 2019-10-23:
Hi Jean,

would it be possible to make some test-data online available. offline.py mentions "lungwgel.bin". Maybe some of these fruit- and glass-scans, if the lung-scan is not possible. And it would be nice to have additional photos of the setup. It makes testing of new reconstruction-algorithms a lot easier. don't have the hardware yet. 
thanks for your work!

JeanRintoul, 2020-01-28:
Hi orgua, Have you received the hardware yet? I have been meaning to do this but have not got around to it yet. If you would like to add this, it'd be great.

ajkh624, 2020-08-31:
HI.
I really want to generate mesh reconstruction image from 'raw data.txt' that I saved from dash board or 'simdata.txt' from simdata.py. Could you show us a demonstration how to generate reconstruction images from simdata.txt using offline.py?

JeanRintoul, 2020-09-02:
please give offline.py another try with the simdata.txt there now. You can also generate your own example data with simdata.py. I've updated the readme with more precise instructions. Thanks.

closed 2020-09-03 by JeanRintoul