OpenEIT/OpenEIT #35: Could not connect

McD-NMI, 2019-11-14:
Running Linux Mint, Python 3.7. 

I  downloaded and installed the OpenEIT repository, ran 
`pip install -r requirements.txt`

had an error about Freetype, followed the recomended command 
`sudo apt-get install freetype6-dev`

Tried agian to install requirements.txt and it works. 

Plug in and turn on the Spectra kit. 
run `python app.py` and go to the http://127.0.0.1:8050/ in either Chrome or Firefox (not working on either).

Press "Connect" and get the following error:

(EIT) matt@matt-MACH-WX9:~/anaconda3/envs/EIT/OpenEIT-master$ python app.py
2019-11-14 17:14:14,297 JAC mesh set up 
completed jac reset
2019-11-14 17:14:15,773 JAC mesh set up 
completed jac reset
d
2019-11-14 17:14:15,861 openeit_server_started
 * Serving Flask app "app" (lazy loading)
 * Environment: development
 * Debug mode: off
/dev/ttyUSB0
connecting to:  /dev/ttyUSB0
could not connect
2019-11-14 17:14:21,242 Cannot connect to /dev/ttyUSB0
problem

JeanRintoul, 2020-01-28:
Question: Does Linux MINT pre-install the VCP FTDI drivers? It's possible that it doesn't which is causing you a problem. 
SPECTRA uses an FTDI chip to communicate via UART. The VCP FTDI driver is not pre-installed on Windows(but is on all other OS's to my knowledge but maybe some variants of Linux are different?!). Here are some instructions relevant to windows for FTDI driver before continuing to the dashboard install: https://www.ftdichip.com/Drivers/VCP.htm . You may need to figure out a way to install or at least check if you have VCP drivers on Linux Mint.

JeanRintoul, 2020-09-03:
Any updates on this issue?

closed 2020-09-03 by JeanRintoul