OpenEIT/OpenEIT #36: .eit file processing

jannisborn, 2020-07-01:
I'm new to EIT but got access to lung EIT data from COVID patients, saved in .eit format. I would like to visualize the data and visually inspect it. I installed the package, but I'm lost on how to start. I experimented with `offline.py` but it doesn't read `.eit` files which are binaries.
Do you have any way of converting `.eit` files to be digestable for OpenEIT? If not, are you aware of any Mac-compatible toolbox/program that can deal `.eit` files? Non freeware like MATLAB is also fine.

Thanks a lot!

JeanRintoul, 2020-07-01:
Hi there, There is not yet a converter for .eit files. I wonder if the software suite EIDORS(matlab) might have a converter? OpenEIT is using a python port of EIDORS called pyEIT - https://github.com/liubenyuan/pyEIT which has a bunch of easy to use examples you may wish to try. https://github.com/liubenyuan/pyEIT/blob/master/doc/pyEIT-data_format.pdf

jannisborn, 2020-07-01:
Dear Jean, thanks for the rapid reply.

Yes, I also came across `pyEIT` but couldn't find any utils for `.eit` processing in there. Since I'm an EIT newbie, so unfortunately not the right person to attempt a converter. 
But I really appreciate the hint to EIDORS, I will install and investigate it in MATLAB. I can keep this issue open for a couple of days and report wheher EIDORS can deal `.eit`.

Have you used/heard about a tool from [Dräger](https://draeger-msi.de/en/services_en/downloads/) to record/process EIT?

JeanRintoul, 2020-07-01:
Great - love to hear about a good .eit parser. 

I think using EIT on Covid patients would be great, particularly MFEIT to determine material obstruction through lungs(i.e. doing EIS at each voxel).

closed 2020-09-03 by JeanRintoul