OpenEIT/OpenEIT #37: Issue with read file - parse_line

nakulpatel2410, 2020-08-14:
I think there is an issue with parse_any_line. I tried to read data from the dashboard and got the error code “Please attach the error you got from terminal, it says something about no module named-parse_line”. I looked through the history of the backend code and saw that parse_line has been updated to parse_any_line. I was wondering if anyone else had come across this issue and if there was a simple fix to resolve it. I think I traced the issue to controller.py-line 86 (see screenshot) where it still calls the old version of the code.

I am a student working on an EIT project to detect muscle contractions of the human stomach. I am still learning to use python and any help would be useful.

[image]

JeanRintoul, 2020-08-14:
A previous version of this code used an earlier library, which is why it changed from parse_line to parse_any_line. Would you be able to compare the precise versions of dash/plotly you have with the requirements document, and confirm that the version numbers match?

nakulpatel2410, 2020-08-18:
Hey Jean thanks for the reply. I have just checked the versions of dash/plotly with the requirements document, and they seem to be the exact same.

JeanRintoul, 2020-09-02:
Would you be able to give step by step instructions to recreate this error?

sam-richardson, 2020-09-03:
I think you probably want controller.py-line 86 to read:

data = OpenEIT.backend.serialhandler.parse_any_line(line,'b')

JeanRintoul, 2020-09-03:
Thx - fixed.

closed 2020-09-03 by JeanRintoul