OpenEIT/OpenEIT #38: no scroll bar

ajkh624, 2020-08-20:
Hello

I connected to spectra, however I cannot perform image analysis because the scroll bar is missing on the image tap.
[image]

please help me with this.
I had uninstalled python and reinstalled everything

JeanRintoul, 2020-08-20:
Please describe how to recreate this error. In the picture I can see a scroll bar. Where is it missing? What operating system, and version of each of the requirements files are you running? How, in step by step form can I recreate this issue? This will help me solve it, otherwise I cannot understand the problem. Thanks.

ajkh624, 2020-08-31:
> Please describe how to recreate this error. In the picture I can see a scroll bar. Where is it missing? What operating system, and version of each of the requirements files are you running? How, in [...]

I meant the sliding bar that is supposed to be under the real time control panel. I reinstalled PIP package in the virtualenv environment and it works fine.

zhjackie2001, 2020-11-05:
I am experiencing the same issue, the scale bar for the intensity is missing, I've updated most of the modules, any idea on which module is responsible for displaying this? 
[image]