OpenEIT/OpenEIT #39: mesh reconstruction image from offline.py

ajkh624, 2020-08-31:
HI.
I really want to generate mesh reconstruction image from 'raw data.txt' that I download from dash board or 'simdata.txt' from simdata.py. Could you show us a demonstration how to generate reconstruction images from simdata.txt using offline.py? 
When I activate offline.py, it mentioned 'lungwgel.bin' missing. If you have a 'lungwgel.bin' file, could you share for me?

JeanRintoul, 2020-09-01:
Hello, I think you should replace with your own file. I will try to look when I have time, but you just need to read in your own file.

ajkh624, 2020-09-02:
I did replace lungwgel.bin with my data file. Here is my data file(rawdata8.txt) and offline_0902.py I ran. 
I used Greit 32 algorithm. Please would you run this if you have a free time? Thank you.

[offline_0902.zip](https://github.com/OpenEIT/OpenEIT/files/5159733/offline_0902.zip)

JeanRintoul, 2020-09-02:
Hello, I just did an update to offline.py to make it more clear. Please give it another go. Another issue is that simdata.txt appends data to the simdata.txt file, so if you change your EIT parameters(i.e. electrode number), it will have data in two different formats- just something to watch out for. 

Here is a reconstructed tomographic image of the datafile you provided using offline.py 
[image]

This new update works for me. Happy to answer any other questions.

closed 2020-09-03 by JeanRintoul