OpenEIT/OpenEIT #45: Problems connecting to device with BigSur

spors, 2021-03-18:
I was not able to establish a connection to the device using the Dashboard. I digged a bit into the problem but was not able to find a solution. I am using a MacBook Pro (2018) with macOS Big Sur 11.2.3

- I plugged the device via the USB cable to the computer. In the system monitor a `FT230X Basic UART` shows up in the USB section. However there is no device created under `/dev`. It seems that the FTDI driver from Apple have some limitations regarding custom product IDs. But there also seem to be no alternative so far. Is there some way to solve this problem?
- I can connect to the device using BT and the Dashboard. Is there some way to debug the connection outside the Dashboard to locate the problem?

JeanRintoul, 2021-03-18:
Have you tried a console program like realterm/coolterm to connect to the device without the dashboard software? If you connect, you should see a stream of UART data come through the port. This would help debug your issue.

spors, 2021-03-18:
Thanks for the suggestion! I tried with CoolTerm connected to the `Bluetooth-Incoming-Port` with its default settings (Baudrate etc.). I couldn't see any data coming through the port :-(

JeanRintoul, 2021-03-18:
The fact it shows up as a UART device suggests it is working but there is another reason you havent been able to connect which may have something to do with anything else on your computer i.e. firewall settings. 
In coolterm, you need to change the port from bluetooth to the particular serial one you want to connect to... and enter correct baudrate and other settings.

spors, 2021-03-18:
There is no other port besides the BT port. Perhaps this is due to the FTDI driver issue mentioned above?I tried it on a different computer with OS X 10.x and there I could see the stream from the USB serial port. However, also on this computer it did not work with BT.

I would like to stick with the newer OS X for practical reasons. Do you have some further ideas how to get BT running?

JeanRintoul, 2021-03-19:
I am using Big Sur too, and it works well for me. In CoolTerm, I can see usbserial-biglongnumberletters turn up as a port option. Have you tried turning the power switch on the other direction on the Spectra when you do this, and re-scanning the ports?

spors, 2021-03-19:
I was able to get the BT connection running under BigSur by

- Allowing the terminal to access to BT in the Security Settings
- Modifying the code related to the OS selection in `OpenEIT/backend/serialhandler.py`. I changed `darwin_xx` to `darwin` as it seems that the former is not reported by `sys.platform` in current Python implementations on OS X

Should I create a pull request with my changes?

The initialization of the BT connection is not very stable and might require some tries. The reason could be that `Adafruit_BluefruitLE` is deprecated https://github.com/adafruit/Adafruit_Python_BluefruitLE.